MagnusPer/Balboa-GS510SZ #1: Testing help

wictorwilen, 2022-09-14:
@MagnusPer - if you need a tester/contributor, please share the code. I'm all set with hardware to test this out!

MagnusPer, 2022-09-14:
Hi there!
First software delivery Phu! Have tested myself a couple of hours and it seems to work as expected so far. Mainly happy case scenario but anyway something is up and running :-)

Try this basic version out and give me some feedback and you are super welcome to contribute.

netmindz, 2022-09-15:
Great work turning this round so quickly

MagnusPer, 2022-09-16:
@netmindz - Thank you! Most of it was actually working with the oscilloscope to understand how the logic was connected. The library has now a limited scope of functionality as a start.

netmindz, 2022-09-16:
You say limited, but from quick look it appeared you had mapped all that the control panel can do.

Only bit that comes to mind for the library would higher level functions like setting a specific temp rather than just +/-

MagnusPer, 2022-09-16:
That is exactly what I working on now to set a specific temp instead of "push" the button x number of times

wictorwilen, 2022-09-16:
If electricity prices is decent this weekend I'll fill up my spa and start testing this! Amazing job so far

MagnusPer, 2022-09-16:
@wictorwilen - Before filling up the SPA with water you might test first the communication working as expected. I can imagine you will have some fault codes in the LCD display such low water etc. But at least you see that it works.

INFO: My top priority on this project is to be able to control when in time during the day the heater should be activated. This in order to save to money which i can spend on something more funnier:-)

netmindz, 2022-09-16:
Yeah that's why I want to get write access to my GL series. Have home assistant put into sleep overnight if the weather looks unsuitable the following evening or weekend day

closed 2022-09-27 by MagnusPer