MagnusPer/Balboa-GS510SZ #2: esphome.io support?

wjcloudy, 2022-10-18:
Hi, thanks for sharing this, going to try this on my own unit!

Just a thought, have you looked at writing this as a custom component in the popular esphome.io ecosystem?
This way the wifi stability, MQTT, HomeAssistant support, OTA, web UI & central management problem is all solved for you!

MagnusPer, 2022-10-19:
Hi @wjcloudy 
Happy you have found this library and please use and provide your feedback. I have used this for a couple of weeks and it serves my main purpose to control when in time the heater should be activated or not.

Your question on using esphome - I have myself not used and by that not have any experience. Feel free to adapt the example code and make a proposal which we can evaluate and provide to the community as an additional example.

wjcloudy, 2022-10-19:
Thanks, will look into it. I also want to use for controlling heater on time. I presume you achieve this by ramping the temp up during periods you want to use power and down at other times?. Have you found a way to adjust the minimum set point?

MagnusPer, 2022-10-19:
Yes - I change the temperature dependent if I want to heat or not. My settings is right now to toggle between 37 and 27 degrees depending on electrical price using the Homey automation Hub

MagnusPer, 2022-12-13:
CLOSED - No activity

closed 2022-12-13 by MagnusPer