MagnusPer/Balboa-GS510SZ #3: VS series compatibility?

kernel610, 2022-10-24:
Any idea if this should work with the VS series Balboa controllers that also use the RJ45? I was about to order the PCB to test it out but wanted to see if you've heard anything from others or foresee any challenges? If not, I can test and provide back an update.

MagnusPer, 2022-10-25:
Hi,
Unfortunately no idea if the Balboa **VS** system is compatible with the GS or not. But after some minutes googling the control system PCB seams at least similar.

As information there are other repositories for other balboa systems, you might can go through those and see if any have asked the same question as you have:
Balboa **GL** -  https://github.com/netmindz/balboa_GL_ML_spa_control
Balboa **BP** -  https://github.com/ccutrer/balboa_worldwide_app

I haven't set up a Discord area for this development, I think we can trace it through issues within this repository so far. But if the community grows, I'll keep that in mind

netmindz, 2022-10-25:
If you look at the manual for at least some of the VS series to see what Topside panels they are compatible with, then look at what control units they work with, I see GS units listed.

What is your exact model of controller and topside panel?

The key thing really is both the GS and GL project are really emulating the panel for their respective controllers

MagnusPer, 2022-10-25:
When reading the topside panel documentation I use for my GS-serie [VL700S](https://github.com/MagnusPer/Balboa-GS510SZ/blob/main/extras/docs/Balboa-Panel-VL700S.pdf) I found here that both GS and VS control system is compatible 

[image]

By that I think this library should work also for the Balboa **VS** serie

kernel610, 2022-10-25:
Controller model is VS501SZ which appears to be covered with the link and screenshot Magnus provided. Topside panel is in rough shape but documentation for the controller shows VL401, VL403, or VL200.

I'm going to order the PCB and give it a try! I'll keep you updated through this thread. Thanks again for the effort put into this!

MagnusPer, 2022-10-25:
Minimum order is 5 PCBs, I have one available which I could send by post to you. Give me your adress by mail magnus_p@live.se

closed 2022-12-23 by MagnusPer