MagnusPer/Balboa-GS510SZ #5: BOM list for PCB?

thoby1, 2022-12-10:
Very interesting project. Thought I would test this, but have limited electronics knowledge.
Could you attach a BOM list so that it would be easier to order all the necessary components for the PCB board?

cbakkerprive, 2022-12-10:
Wow, cool! I too am definitely interested in the BOM list.

MagnusPer, 2022-12-10:
I have been using this for a couple of months now and it works as I expected with my main purpose to heat the SPA during the cheapest hours.

The BOM list hopefully explained in the [Wiring](https://github.com/MagnusPer/Balboa-GS510SZ#wiring) section. What you need is the following:

3 pcs  - 1 kOhm resistor
2 pcs  - 2 kOhm resistor
1 pcs  - 220 Ohm resistor
1 pcs  - 370 ohm resistor
2 pcs -  EL816 optocoupler 
1 pcs - Wemos D1 mini 
1 pcs - RJ45 cable

thoby1, 2022-12-10:
Cool to hear it works! The use case with turning the heater on and off during low and high prices was exactly what I had in mind.

When it comes to purchase of parts there seems to be so much extra specifications that makes it hard to choose. For resistors for example there are spec for Voltage and Watt. Take the 2kOhm for example, will this item work:
**NQ06W4F2001T5E**
[](https://jlcpcb.com/partdetail/1050181-NQ06W4F2001T5E/C965049)

MagnusPer, 2022-12-10:
I uploaded the BOM list [PCB layout](https://github.com/MagnusPer/Balboa-GS510SZ/tree/main/extras/PCB%20layout) I used when I purchased the PCB, hope this helps

cbakkerprive, 2022-12-10:
Thank you!
Can you also provide the CPL file so I can order assembled from JLCPCB?

MagnusPer, 2022-12-10:
@cbakkerprive - Not sure what you mean with the CPL file but I guess its the [PickAndPlace](https://github.com/MagnusPer/Balboa-GS510SZ/blob/main/extras/PCB%20layout/PickAndPlace_PCB_Balboa-%20GS510SZ_2022-12-10.csv) file?

cbakkerprive, 2022-12-11:
Thanks MagnusPer!
Indeed I did the PickAndPlace File.
As stated I try to order the assembled PCB from JLCPCB.
I did uploaded the 3 files to JLCPCB, but unfortunately after selecting the BOM and the PickAndPlace file (and select Next) I get the following error:

_Incorrect format
File parsing timeout. Please try again later.__

Maybe I should try to open the gerber file in the EasyEDA designer myself.
Thanks anyway.

MagnusPer, 2022-12-11:
@cbakkerprive - I think it's better you try yourself to create the PCB in e.g. EasyEda with the help of the Gerber files provided. I remember I needed to replace and change the optocouplers component due to missing parts in the factory. When you succeed to place the order please help others here by attaching your PCB production files.

If you have any questions I'm here to help the best I can!

thoby1, 2022-12-30:
I havesuccessfuly completed a order at JLCPCB with the "BOM.xls" and "PickAndPlace_PCB_Balboa-GS510SZ_2022-12-10.csv" file. This issue can be colsed.

closed 2022-12-30 by thoby1