MagnusPer/Balboa-GS510SZ #6: Support for VL600S panel?

thoby1, 2022-12-30:
I tested the software against a VL600S panel. It seems to give the correct status messages, but when I try to send commands the display stops working. Then only a gray background is displayed without any text. When this happens, I have to turn off the power and disconnect the circuit board with the Arduino. Then when I turn on the power again with the original setup, the display works fine. Have tried the "stop" and "reset" function, but it has no effect on the issue.

MagnusPer, 2022-12-30:
To start somewhere...
If you press the display button when the circuit board is connected, does the e.g. jets starts or temperature change? by this I want to understand if the connection is OK between the controller unit and the display with the circuit board connected.

thoby1, 2022-12-30:
I am using the MQTT example.

Looks like the same thing happens when you press the control panel. Also managed to provoke the error by just waiting a few minutes without pressing a single button or sending any command. I think you are on to something about there not being a good connection between the control unit and the display, when the circuit board is connected. I have tested the TP cables with a network cable tester and they worked fine. Also tested the circuit board with the same cable tester. There I got connection on all pins except for pin number 3.

MagnusPer, 2022-12-31:
Yes - something is wrong since you not manage to press the display button. 

Have you used ordinary TP cables? They normally have crossed lines! 

You can not measure line 3 since the optocoupler requires 5V to close. you could provoke that by manually add 5V and GND on the bench to check

thoby1, 2023-01-01:
- I’m using the original flat cable that comes from the display and my own twisted pair cat 6 straight cable to the control system. The cables has a straight connection on etch pin. Should I use only flat cables and not the one that has twisted pairs? The two cables work together if I disconnect the circuit board and connect them with a RJ45 cable extender.

- Tested the optocoupler by giving it 5V from line 3 on RJ1. That gave me a signal on line 3 from RJ2.

- I have wired the display the RJ1 connector (it is marked with “Display” on the circuit board) and the control system to RJ2.

I see that I am using version 3 of the Wemos D1 mini, but in your wiring diagram it shows version 4. Will that be a problem?

To be more specific in the issue. The system seems to be working fine during the hot tub startup sequence when I turn the fuse on. Then I get MQTT messages with water temperature and pump status etc. After the startup the display begins to flash between showing the water temperature and just a gray background. Eventually just ending up in showing a gray background. This happens regardless of me sending any commands or not. To correct the issue I have to put everything back to the origianl setup. Since the cables work fine with just a cable extender and not on the circuit board could there be too much resistance on some of the pins?

MagnusPer, 2023-01-02:
Seems the TP-cables are ok when a RJ45 cable extender is used. I think it's more likely something with the circuit board. If we suspect the Wemos then you can disconnect that from the board since the board should work without it, check the [wiring diagram](https://github.com/MagnusPer/Balboa-GS510SZ#wiring).

thoby1, 2023-01-02:
Tested a circuit board without Wemos and it worked fine. All buttons and display worked as they should.

Should the Wemos have an external power supply? While testing I only have only connected the two TP cables. They have 5V on pin 7 and ground on pin 4. I was wondering if the Wemos draws too much power.

MagnusPer, 2023-01-02:
Ok, we are closing in! 
To be sure about the Wemos you are using. Is this the ESP8266 or the newer ESP32 version? 
A guess is the Wemos restarts and might cause the problem. To check this connect your computer an log via the serial interface to check what’s happening

thoby1, 2023-01-04:
I have now tested the Balboa Read example software and connected it to a PC for debugging.

It worked fine from the software side. I saw the status and could start and stop the pump, turn lights on and off, etc.

When I pressed the display, it behaved strangely. It didn't always react and if it did it seemed as if the buttons had been given a different function. If I pressed “Pump 1” it looked like I pressed “Temp up”.
Finally, I ended up with the same error. Showing a gray background in the displayed again.

The difference in this test scenario is that Wemos is supplied with external power from the USB cable and that this software do not use the WiFi module. So maybe there is something with the power?

Also, if I used the board “LOLIN(WeMos) D1 R1” the software did not work. I ended up using the “LOLIN(WEMOS) D1 R2 & mini” board config.

MagnusPer, 2023-01-04:
I just thinking about the VL600S (6 button) panel and the difference compared to the VL700S (7 button) I have - when reading the [Balboa-GS510SZ](https://github.com/MagnusPer/Balboa-GS510SZ/blob/main/extras/docs/Balboa-GS510SZ.pdf) documentation there are some jumper settings to configure the control system with depending on panel configuration. And if this effects the way the commands is sent??

> It worked fine from the software side. I saw the status and could start and stop the pump, turn lights on and off, etc.

Need to double check this, do you mean when entering the write commands using the serial monitor the the activations is correct?

> When I pressed the display, it behaved strangely.   

This is really strange since the signals should go straight through the optocoupler - see the [wiring diagram](https://github.com/MagnusPer/Balboa-GS510SZ#wiring). Like when the Wemos was dismounted.  

Did you order more then one circuit board, have you tried another one?

thoby1, 2023-01-07:
I have limied knowlage about this but I think there is something wrong with the circuit board. When I tested with voltage on the Wemos, it outputs 5V on pin 7 and 4 (ground) (RJ2). If I did the opposite and put voltage on pin 7 and 4 at RJ2, I only got about 1V on pin 8 and 7 on the Wemos. Suspect that there may be something wrong with the common ground somehow. Because if I tested pin 8 on Wemos against pin 4 on RJ2, I got 5V. Tried different places where I had access to common ground such as on one side of R3 I also only got 1V.

When it comes to the jumper settings on the motherboard such as J12, I interpret it as which software should run. I use this setting:

**Jumper must be on Pins 1 and 2 for GS51xZ/GS52xZ/GS5xxSZ/GS5xxDZ software.**

I have a spare circuit board that I have tested without Wemos and it works fine. Have ordered a new Wemos v4.0 now. Will test again when I get it. I believe that it will work then, because when I tested with the Read example and sent commands via the serial interface, I was able to set the temperature, turn the pump on and off and turn on the light. The Wemos never seemed to restart.

MagnusPer, 2023-01-07:
Tricky when I not myself can measure the circuit board. What I at least have done is to update the [wiring diagram](https://github.com/MagnusPer/Balboa-GS510SZ/blob/main/README.md#wiring) to clarify the Vcc +5V and ground connections.  

You are on the correct track, verifying piece per piece, at least we have learned and have zoomed in where the problem is located. If possible, please take a picture on your board with the Wemos mounted and paste it here.

Give me an update when you have tested with the new Wemos.

thoby1, 2023-01-13:
I got the same error now with a new circuit board with v.4.0 of Wemos. My problem is that I don't get enough voltage to the Wemos from RJ1 or RJ2. My test setup is a USB mobile charger which provides 5v. It works if I connect the USB to the Wemos, but if I apply voltage to pins 7 and 4 on RJ1 or RJ2, I only get about 1v on the Wemos. I have soldered all the pins from the Wemos firmly to the circuit board, although from the drawing only pins 3, 4, 7, 8 and 15 needs to be connected. Will that be a problem? Holes have been made for all the pins on the circuit board, so I assumed that the ones that are not in use are isolated.

Have attached a picture of the first circuit board. The soldering doesn't look so good after I've tried soldering the Wemos off and on again.

[image]

MagnusPer, 2023-01-13:
The unused pins is not connected to anything, no problems to solder them. 

When measure with a standard multimeter it takes the average of the voltage, since it is a communication line you will never have 5V since this is the peak value. 

I need to do my own bench test setup to guide you further. I will be back home on Sunday, so expert som days until I have some data to provide. 

But until then - I tried to zoom in on the Wemos board to look after some data, can you read ESP8266 or ESP32 on the chip?

thoby1, 2023-01-24:
Installed version 4 of Wemos LOLIN D1 mini [https://www.wemos.cc/en/latest/d1/d1_mini.html](url) on a diffrent circuit board and now it is working as expected. Guess I was just unlucky with either my old Wemos or the circuit board. Regardless. The VL600S display panel can also be used with this project.

closed 2023-01-24 by thoby1

thoby1, 2026-03-28:
It works partially for me. I only have one pump, so I haven’t experienced that specific behavior.

However, there does seem to be an issue with the panel. If I operate everything from Home Assistant only, it stays stable over longer periods. But when I start using the physical panel, things seem to get out of sync — especially when adjusting the temperature (Temp Up/Down causes the most trouble).

My guess is that our panels might have slight differences in timing between signals. Eventually, it gets to a point where none of the buttons respond anymore, and I have to power cycle the system to recover.