MagnusPer/Balboa-GS510SZ #7: Compability GS501Z and VL200

richardtjena, 2023-02-28:
Hi!

I have a Balboa GS501Z-control system with VL200 duplex display which uses RJ45 but only have four buttons.

Do we know if the data protocol is the same regardless if the display is a Duplex- (GS5xxZ), Standard- (GS5xxSZ), or Deluxepanel (GS5xxDZ-unit)?

BR

MagnusPer, 2023-02-28:
@richardtjena  - Welcome in here to this repository!

When reading the specs of the control system [GS501SZ](https://www.balboawatergroup.com/getdoc.cfm?id=2160) and [GS510Z](https://www.balboawatergroup.com/getdoc.cfm?id=2169) there are some differences as you mentioned where the VL200 panel has less buttons.

To be honest, I have no idea if this will work or not, but my guess based on the documentation is that it should work as both features like the circuit board layout seem to be pretty similar.

The best answer is if you make a try - The hardware cost is where low!

MagnusPer, 2023-03-03:
Closing this issue since this repository does not support Balboa GS501Z-control system with VL200 panel

closed 2023-03-03 by MagnusPer

richardtjena, 2023-04-10:
@MagnusPer was it a user with a public repo? Any chance you can point me to his direction.

I also thought about flipping the J12 jumper in the control unit but don't know if it would have any effect without changing the ROM.

StarkJohan, 2024-04-15:
I have the same setup and the protocols must be very similar. To avoid doing the work someone else already has done, please put me in contact with anyone that might help: johan@punktvvv.se

netmindz, 2024-04-18:
The way to know about compatibility is to think about the fact we are emulating a panel, so if you look at which official panels are compatible with which controllers you know that they must use the same communication

StarkJohan, 2024-04-21:
I'm using the VL240 panel and the protocol is indeed similar but there's a few big differences. I've outlined them in another issue.
In short it will not be possible to use this repo or the hardware without rather big changes.