MagnusPer/Balboa-GS510SZ #8: Confirming ground pin?

techoguy, 2023-05-07:
Hi, just curious how you were able to confirm which pin was the ground. I'm currently sniffing my spa panel bus and it seems to have a similar protocol (though different) compared to yours. Mine has only 3 groups of 7 data and handles the buttons the same way as yours. But I'm trying to confirm which pin is the ground pin. 

My protocol has a different odd case when the clock is held high until the next batch of data. Still trying to figure out why it does that (it might be indicating the heater on LED on the panel, but not sure yet). 

I'm still annalizing my data to see how I want to implement this but seems like I might be able to use yours as a baseline. Thanks for the work you've already put into this.

MagnusPer, 2023-05-08:
Hi,
Great this repository can advice and help you with your project. The work I did was more based on "exclusion method" where I finally understood the SPA behavior and the usage of each pin. To find the ground pin you can measure against the control unit ground which should have the same.

closed 2023-08-06 by MagnusPer

netmindz, 2023-08-07:
What model do you have @techoguy ?

One way to confirm ground is with the power turned off use a multi meter to check for continuity between that pin and the ground plane of the PCB

techoguy, 2023-08-09:
@netmindz I have a Sundance Camden with 780 Series I-Touch Controls. I think it's about 10-15 years old. Below is a picture of the current signals.

- there is an 8 pin connector going to the control panel and an empty RJ45 plug
- Pins 1-6 are the same between both connectors, but pins 7 and 8 are not
- the control board sends a sequence every 100ms (3 bytes of data to the touch panel, 7 bits from the touch panel for button presses)

**Pins:**

1. 5v
2. GND
3.  Data
4.  Clock
5.  Select (active low)
6. Button Data
7. (Not sure) maybe a device select? 
8. (Not sure) maybe a device select? 

Signal Example: 
[image]

Picture of control board connectors: 
[image]