MagnusPer/Balboa-GS510SZ #10: HowTo

duczz, 2023-09-02:
Hey, i have this baboa Whirlpool too and wanna try these, but iam missing a HowTo.
Currently i have flashed your Project to a wemos Mini (with my settings) but i have no entitied in the homeassistant mqtt server.

What are ne next steps? Docker witch bwa?

netmindz, 2023-09-02:
Have you tried the simple read example first to confirm the basic setup works first?

You don't mention if you have any existing Home Assistant setup already on place. You will need that and follow an appropriate guide to setup MQTT with Home Assistant

There are many different ways for this depending on how/where you are running HA

Once that's setup, you should see the spa under MQTT inside HA

MagnusPer, 2023-09-02:
Hi and happy that you are interested in this library.

Firstly, Have you set your credentials in the code for the MQTT broker?
[image]

If so check that MQTT messages coming in from the SPA using a MQTT tool like http://mqtt-explorer.com/

duczz, 2023-09-02:
Thanks for reply guys.
I have a MQTT Integration on my homeassitant, these i have enters already in the config, thats why i see via MQTT Explorer the Spa Parameters, but not on homeasssitant
[image]

MagnusPer, 2023-09-02:
Sorry, I do not have HA instead I use Homey so the integration to HA using MQTT I can not help you with

duczz, 2023-09-02:
okay thank, so the next steps are only to let create the PCB layout and than put the j1 in the new medoul and j2 from the modul to j1 from the balboa right? not more to do?

netmindz, 2023-09-02:
You could use with HA with just the raw MQTT, but it's *much* easier if you have a proper device registered in HA

netmindz, 2023-09-02:
Ok, that's the basic read-only example up there now

[image]

duczz, 2023-09-02:
Perfect, the other will follow too?

[image]

netmindz, 2023-09-02:
[image]

netmindz, 2023-09-02:
Latest version with buttons too - untested as I don't own that tub

duczz, 2023-09-05:
to close the ticket, can someone answer my question?
so i need to let create the PCB layout and than put the j1 in the new modul and j2 from the modul back to j1 from the balboa right, than im finished?

can i control than the modies too and set the time for filter?

MagnusPer, 2023-09-06:
@duczz - Between the display and the control system there is a cable with an RJ45 connector. Disconnect that from the Control system and place PCB in between. You need to create/buy the cable with RJ45 connectors between J2 -> Control system, NOTE that should have straight lines and not crossed as in a normal TP-cable.     

**DISPLAY**-------->(J1)  **PCB**  (J2)--------->**Control system**

There are no functions created to change modes or set filter times etc. The library however supports buttons to be pressed so theoretically it should be possible - see the section [button data ](https://github.com/MagnusPer/Balboa-GS510SZ#button-data)  

I hope this mini guide will help you to get it up and running

duczz, 2023-09-06:
Perfect thank you Magnus.
@netmindz can you add all Buttons on this display to homeassitant? Than i can create automatisations to set mode / filtertimes etc.

[image]

netmindz, 2023-09-06:
I won't be adding any more to the code, if the buttons are supported by the library then it should be easy to follow the current examples.. If not, then update the library part of the code too

Create a fork then and then submit a PR

duczz, 2023-09-06:
@netmindz alright, thanks for the first homeAsssitant port steps.
@MagnusPer are you able to update these? it looks like all buttons would be exist. i'm to noobie to port these :(
[image]

MagnusPer, 2023-09-06:
Sorry, I don’t have the knowledge how the Home Assistant works. The library support the “Mode” button and what I see is included in the HA example. The only missing piece is to add a HA button to trigger the mode.

knogler, 2023-10-21:
@duczz have you found a solution?

netmindz, 2023-10-21:
To add a button for pump3, just copy and paste what's listed for pump2

```
pump2.setName("Pump2");
pump2Button.setName("Pump2");
pump2Button.onCommand(onButtonPress);

```

but name pump3. Do same for any other buttons you want such as Mode

netmindz, 2023-10-21:
As well as the definition of he button 

`HAButton pump2Button("Pump2");`

knogler, 2023-10-21:
Thanks for the answer, I probably could have done that.

Unfortunately, setting the temperature is not displayed. somehow the variable “setTemperature” is missing from the homeassist file. And unfortunately I don't know how to create this in order to be able to regulate it.

duczz, 2023-10-25:
@knogler nope, i was not able to migrate it complet to HomeAsssitant. I think this will never be done ^^ (i'm no developer :/)

netmindz, 2023-10-30:
Try this version @knogler  https://github.com/MagnusPer/Balboa-GS510SZ/pull/13

duczz, 2023-11-09:
@knogler Could you test it successfully?

J0E-Cool, 2023-11-15:
Hi @all,
just a dumb question ;)  and sorry to reuse this chat...
I've already built the circuit on a stripboard 🙈, but now I'm thinking about, how I could test it with the still installed display VL100 and control board GS100... because, I guess, I've to adapt the programming.
Do you have some tips, lessons learned, how you did/do that? 
Thanks :)

knogler, 2023-11-17:
Sorry for my late response. and thanks for coding. Unfortunately I haven't had time to test yet. I also have to see how I can get the software modified so that it runs with Balboa-MGS504DZ with panel VL700S (MAY800D). As soon as I have more information, I'll let you know.

netmindz, 2023-11-17:
The model of top panel shouldn't really matter too much as we are emulating a panel. So what matters most is compatibility with the controller and the protocol it's speaking. Different models in the same range should use the same protocol so it shouldn't need code change.

What you might find is that your tub might only have 1 pump not two, so you mind find that pressuring pump2 might trigger the blower, just as if you had the wrong overlay applied to your top panel for your particular configuration

MagnusPer, 2023-11-17:
> Hi @ALL, just a dumb question ;) and sorry to reuse this chat... I've already built the circuit on a stripboard 🙈, but now I'm thinking about, how I could test it with the still installed display [...]

[@JOE-cool](https://github.com/J0E-Cool)

1) Start to just connect the circuit board to see everything works by manually press the panel buttons.
2) Easiest is to use example code [Balboa_GS_Read](https://github.com/MagnusPer/Balboa-GS510SZ/tree/main/examples/Balboa_GS_Read) and monitor via the serial monitor in [Arduino IDE ](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-serial-monitor) if data is observed.
3) You can also use the example code via the serial monitor input to activate button presses according to the [code](https://github.com/MagnusPer/Balboa-GS510SZ/blob/main/examples/Balboa_GS_Read/Balboa_GS_Read.ino). Example is `4` to activate light 
4) Now its time for remote access and automation. Depending of your knowledge and needs you have either the MQTT or HomeAssistant to work with

Good luck!

J0E-Cool, 2023-11-17:
@MagnusPer thank you, helps me a lot :) I hope I can continue this weekend, if the weather allows. 
For the automation part, I've still to decide which one would fit best (don't have any in place), but step-by-step ;)

ggtimtom, 2024-01-05:
Hello, i also try to make my GS523DZ with Panel VL802DZ smart. I hope the protocol will be similar.
I will start when the weather is fine.

malu254, 2024-01-05:
> Hello, i also try to make my GS523DZ with Panel VL802DZ smart. I hope the protocol will be similar. I will start when the weather is fine.

Hello,
I think you need the Version from here https://github.com/Shuraxxx/-Balboa-GS523SZ-with-panel-VL801D-DeluxeSerie--MQTT 
The Hardware which you need will be the same

duczz, 2024-01-13:
i try to build curretnly my lovelance Layout for the Summer. :)
Is it possbile to switch all Buttons with on/off function into a switch? what i need to edit for that?

MagnusPer, 2024-01-13:
I using NodeRed as dashboard for my home automation. In NodeRed I create buttons triggering the MQTT commands and subscribe to MQTT status. Simular should be possible to do using Lovelance.   

[image]

[image]

duczz, 2024-01-13:
Thanksmagnus, I had hoped that we could change habutton to haswitch in the *.ino File direct but i dont know If this works. Maybe @netmindz know it? :) (i dont know If haswitch exists in this program language)

netmindz, 2024-01-13:
Yes there is @duczz  - see https://github.com/netmindz/balboa_GL_ML_spa_control/blob/7e193eb2708c2803dda82cae5e14056a8bfcb445/sensor/src/sensor.cpp#L120

The key thing is if you aren't just a button, then you need to know the current state, so it's not just swapping the type, but still pretty simple to do. See standalone example https://github.com/dawidchyrzynski/arduino-home-assistant/blob/main/examples/light/light.ino

netmindz, 2024-01-13:
I haven't tested the code, but it will be something along the lines of this https://github.com/netmindz/Balboa-GS510SZ/commit/4d32135e64c111bf3d1619eb5a832677b0603a93

duczz, 2024-01-14:
i have created on netmindz top tipps and example (much thanks without i cant do this) a new version, magnus can you create a new "Balboa_GS_HomeAssistant_DEV.ino" with this content? Would be create when someone can test it, my platine is not delivered :(
https://pastebin.com/7bMn9JAM

Someone knows why line 237 is commented out?

netmindz, 2024-01-14:
@duczz  - I'm not familiar with this specific controller, but I know on some spas the buttons have multiple functions, so even if there are switches for the pumps, it might not be wise also to remove the ability to send button presses, unless you know for sure that you only need mode and temp +/-

As for your onSwitchCommand, this needs to look to see **which** switch you are trying to change and then read the right state, you currently are reading the state of the light.

For example of checking the sender see https://github.com/netmindz/balboa_GL_ML_spa_control/blob/7e193eb2708c2803dda82cae5e14056a8bfcb445/sensor/src/sensor.cpp#L240

MagnusPer, 2024-01-14:
@duczz - When you have a working and tested example you can do a Pull Request and then I can approve it to be added

duczz, 2024-01-17:
okay thanks netminz, i tried, but that programming is to high for me im out ^^

awallace24335, 2024-01-27:
> @duczz - Between the display and the control system there is a cable with an RJ45 connector. Disconnect that from the Control system and place PCB in between. You need to create/buy the cable with [...]

Just to confirm, can this be a normal patch cable or do the pairs have to be untwisted? 

Thank you

knogler, 2024-01-28:
yes, it's a normal patch cable

jarnodrenth, 2024-11-24:
> Sorry, I do not have HA instead I use Homey so the integration to HA using MQTT I can not help you with

 Can you tell me what you use for homey?