MagnusPer/Balboa-GS510SZ #14: No possibility to buy the PCB via JLPCB

malu254, 2023-11-12:
Hello @MagnusPer , do you have any of your circuit boards for the Balboa control system left? Unfortunately, the Everlight Elec EL816S1(D)(TA)-F is currently not available for purchase via JLPCB. I already have the WemosD1Mini, so I would only need the board with the soldered SMD components.
If you still have a circuit board, how much would it cost including shipping to Germany?

Sorry for opening an issue, but Github has unfortunately removed the private message tab.

BR 
malu254

MagnusPer, 2023-11-12:
@malu254  - Unfortunately I have no extra circuit board. But the JLPCB must be able to give you a proposal of a similar one since that optocoupler is very regular.

malu254, 2023-11-12:
Ok, too bad, unfortunately JLPCB can't suggest any others, or am I missing something?
[image]

Sh4d, 2023-11-18:
You can order them from China for relatively cheap - https://www.aliexpress.com/item/1005004701331738.html?spm=a2g0o.order_list.order_list_main.71.48c918022AP3k2

I got mine to Canada in about 2 weeks.

malu254, 2023-11-26:
@MagnusPer @Sh4d thanks for your Help. I managed it to switch the Optocoupplers to similar ones

closed 2023-11-26 by malu254

kernel610, 2024-01-12:
> @MagnusPer @Sh4d thanks for your Help. I managed it to switch the Optocoupplers to similar ones

@malu254 What did you choose as the replacement that worked? I'm placing an order for another built board since my board shorted out from water intrusion in the box. I'm seeing the same stock shortage message on the octocoupler.

malu254, 2024-01-12:
> > @MagnusPer @Sh4d thanks for your Help. I managed it to switch the Optocoupplers to similar ones
> 
> @malu254 What did you choose as the replacement that worked? I'm placing an order for another built board since my board shorted out from water intrusion in the box. I'm seeing the same stock shortage message on the optocoupler.

@kernel610 i used this ones: TLP785(BL-TP6,F(C

But if you want i would have to spare PCBs to Sell. They would be completely finished except for the D1 mini Micro Controller

PerssonNiklas, 2024-07-29:
> > > @MagnusPer @Sh4d thanks for your Help. I managed it to switch the Optocoupplers to similar ones
> > 
> > 
> > @malu254 What did you choose as the replacement that worked? I'm placing an order for another built board since my board shorted out from water intrusion in the box. I'm seeing the same stock shortage message on the optocoupler.
> 
> @kernel610 i used this ones: TLP785(BL-TP6,F(C
> 
> But if you want i would have two spare PCBs to Sell. They would be completely finished except for the D1 mini Micro Controller

You dont happend to have one left for sale perhaps? Old thread, worth a shot!

colsonpete, 2024-07-29:
I have a couple of the base boards (no D1 Mini) going spare if you are interested? 

I haven't got one working yet, but suspect that's down to me trying to to get it to work between a GS515Z controller and a VL406T panel.

PerssonNiklas, 2024-07-30:
> I have a couple of the base boards (no D1 Mini) going spare if you are interested?
> 
> I haven't got one working yet, but suspect that's down to me trying to to get it to work between a GS515Z controller and a VL406T panel.

Whats your email so we can get in touch?

colsonpete, 2024-07-30:
Hi NiklasMy email is peter Dot colson at pc-ss dot co dot ukKind regardsPeter Colson

view it on GitHub, or 
unsubscribe.