MagnusPer/Balboa-GS510SZ #16: What about the second connector on the board

Zetto12, 2023-12-10:
Hi Guys,

First of thank you for publishing this project. My Spa controller (Balboa CPGS501Z) has two identical RJ45 connectors. One the Display is plugged in but the other one is empty.

Does the D1 Mini need to go in line with the display or can it be connected directly to the second port? I believe that is where the original WIFI controller is connected to.

Thank you

MagnusPer, 2023-12-10:
@Zetto12 - During the development I tested the "empty" RJ45 connector but it didn't worked and decided to not spend much investigation on why.

closed 2024-01-13 by MagnusPer

StarkJohan, 2024-04-15:
The secondary port is used for an IR receiver like this: https://www.tubs.parts/en/ir-receiver-for-balboa-gs5-systems-p-1509.html (two versions depending on the hardware/software version). 
It would probably be possible to use that input but you'd have to figure out how the IR signals interact with the board. One idea I had was that the signal from the IR receiver might be using pin 1+2. That's a complete guess though. 
Does the display work in both J1 and J2? I would assume so but I haven't tested.