MagnusPer/Balboa-GS510SZ #17: Gs514sz with vl702s topside hasn’t got full functionality

awallace24335, 2024-01-26:
Hello and thank you so much for this project. 

I have set everything up using the HA example and 1/2 of the functions seem to work fine and the rest don’t. 

The functions that are not functioning are:
Mode button
Temp up
Temp down
Lights button + lights 
[image]
on data

Heater status, pump 1, pump 2, display data and temp all work fine. 

My topside is slightly different being the Vl702s so wondering if there could be a slight variation in  the coding.

is there anything I can try to do, to get this fully functioning. 

Many thanks

awallace24335, 2024-01-26:
[image]

MagnusPer, 2024-01-27:
Since that display is different than I developed the code for you need to go back to the basis. The only way I figured out was to study the communication behaviour between the control unit and the display using oscilloscope to determine and mimic each button push.

MagnusPer, 2024-01-27:
If you using the https://github.com/MagnusPer/Balboa-GS510SZ/tree/main/examples/Balboa_GS_Read example and try increasing the temp by enter  “1” in serial consol does anything change on display?

awallace24335, 2024-01-28:
Thank you for your comment,
 after doing some investigating most of the functions that a written through the serial console work as expected. the only problem is when I press "7" it activates the lights and pressing "4" has no response. How would I change these functions around?

I know just need to plug the board back in with the Home Assistant  firmware on to see if that works as expected.
 
Thank you for your help.

netmindz, 2024-01-28:
Are you using the MQTT example or this one examples/Balboa_GS_HomeAssistant/Balboa_GS_HomeAssistant.ino @awallace24335 ?

netmindz, 2024-01-28:
Try adding a space in the name here so the command mapping matches the name of this button.
I don't own this tub so wasn't able to fully test myself

https://github.com/MagnusPer/Balboa-GS510SZ/blob/46f41fbbb927061c465f90602a6bfaaeec2079b3/examples/Balboa_GS_HomeAssistant/Balboa_GS_HomeAssistant.ino#L252

awallace24335, 2024-01-28:
> Are you using the MQTT example or this one examples/Balboa_GS_HomeAssistant/Balboa_GS_HomeAssistant.ino @awallace24335 ?

I was trying to use the Home assistant example but I could get the temp commands working. I then started from the start and used the read example which was working other than lights / pump3 was mixed up. 

I have just tried the Mqtt example and I can get everything working other that the lights / pump3 being mixed up, so i thing there must be a problem in the home assistant example

netmindz, 2024-01-28:
Yeah there is a fault in the home assistant example, the line I listed should be "Temp Up” not "TempUp" and similar for down

awallace24335, 2024-01-28:
> Try adding a space in the name here so the command mapping matches the name of this button. I don't own this tub so wasn't able to fully test myself [...]

adding this space on both up and down fixed this issue of the Temp Up and Temp Down commands. The only thing that I am missing is the Climate option (temp) I don't think its receiving all the values.

[image]

[image]

Thank you

awallace24335, 2024-01-28:
> Yeah there is a fault in the home assistant example, the line I listed should be "Temp Up” not "TempUp" and similar for down

I'm not really sure fully how I fixed it but all seems to be working. I did make some alterations to the code.

[image]

https://pastebin.com/xWgw1sWf

netmindz, 2024-01-29:
The current and target match between the climate and sensors in that image, so I'm guessing you mean the spa itself hasn't changed?

awallace24335, 2024-01-29:
> The current and target match between the climate and sensors in that image, so I'm guessing you mean the spa itself hasn't changed?

Correct, the only reason the target temp works is because i added "hvac.setTargetTemperature(Balboa.setTemperature);" which i now believe is wrong.
to fix the water temp i changed "hvac.setCurrentTargetTemperature(Balboa.waterTemperature);" to "hvac.setCurrentTemperature(Balboa.waterTemperature);"

when i press pump 3 it activates the lights, do you know if this can be switched around?

thank you for your help

netmindz, 2024-01-30:
Clearly different outputs have been used on your controller to the author of the library

Just swap this line

else if (s_payload == "Pump3") {

With the line about light

awallace24335, 2024-01-31:
> Clearly different outputs have been used on your controller to the author of the library
> 
> Just swap this line
> 
> else if (s_payload == "Pump3") {
> 
> With the line about light

No problem i will do that. Thank you

I don't think the status is tracked of Pump 3 to verify if its on or not. Is it possible to add it here or will it not work?

[image]

netmindz, 2024-04-18:
I've created a PR with some untested code that might help you along https://github.com/MagnusPer/Balboa-GS510SZ/pull/19