MagnusPer/Balboa-GS510SZ #18: Hello, I don't suppose anyone in the UK could either offer for sale the PCB including or gates, or give clear instructions how to purchase the made up PCB including components from an online UK PCB maker. I'm struggling with this, many thanks for any pointers (for MS500ZE unit in Masterspa GL2) with RJ45 socket).

doctorphilt, 2024-04-13:


netmindz, 2024-04-13:
Do you know that your board is actually a Balboa control system, just by another model number or is that a typo in your post? 

Just the use of an RJ45 connector isn't sufficient to know compatibility

doctorphilt, 2024-04-14:
Thanks for getting back to me. Yes it is definitely a Balboa. It full "agency model" number is GS5-MS500ZE-RCA-2.0K. It has a standard Balboa GS casing.

StarkJohan, 2024-04-15:
I started decoding the protocol about four years ago but neither had the time or the grit to get it sorted and had to shelve it at that point. I just discovered this project and my aim is to get it up and running in before summer hits here in Sweden. I intend to order a PCB and as MOQ is 10 pcs I will likely have a few spare ones. I will make no promises at this point but I know my way around PCB design and this is a very basic board. I just need to have closer look at the details.

doctorphilt, 2024-04-15:
That would be wonderful if you had one spare / for sale 😃 Not sure we would communicate that if /when happened?

StarkJohan, 2024-04-21:
Unfortunately the protocol is different as well as the pinout. 5VDC is on pin 1. GND, display and clock are the same but I'm yet to find the button data. The other three pins are some kind of low voltage copy of the display data about 1.5V max representing a 1 and approximately half that representing a 0.
Also, there's 3x7bits + 1x3bits on the display line. Collecting data now to see if I can figure it out.
So far all I know is that the middle bit of the last 3-bit chunk is the ligjhts...

StarkJohan, 2024-06-09:
I've ordered a bare PCB that should be compatible with the following controllers and their equivalents:
"Mini": VL200, VL240, VL260, VL400, VL401, VL402, VL403, VL404, VL406T, VL406U
"Deluxe": VL600S, VL601S, VL700S, VL701S, VL702S

The pinout, the protocol and the exact functions differ a bit between the two groups. 
The pinout is easily adjusted with a couple of jumpers in my design. 
The software needs a slight adjustment for the "deluxe" panels but rather large changes to be compatible with the "mini" models. I've got a POC running nicely since a few weeks and I will publish a repo as soon as I've had the time to tidy up everything a bit. 

I will be populating a few of the PCBs in the coming month so if anyone is keen on testing/developing on any other panel than VL406U or VL700S, please let me know below. If everything checks out I will be open sourcing the hardware design as well as the altered code I'm using.

doctorphilt, 2024-06-09:
Hi again, I think mine is VL400 but I would be very keen to trial one (within my limited electronics ability) or purchase one...

StarkJohan, 2024-06-09:
VL400 has one button for temp, mine has up and down buttons for temps so while the physical input is very likely to be the same, the exact function might differ slightly and thus the software would need to be slightly different. If you are a bit familiar with the software side of things you should be able to adjust the code accordingly.

Exactly what is the procedure to change temp from your VL400 controller?

On my controller the current temp is showed by default. If I push up or down once the set temperature starts flashing slowly. Then for every push up/down the set temp changes 1 degree. After a short timeout the flashing stops and the current temp is shown.

I can't offer any PCBs for free but I'll be offering this run at a price that covers my costs.

doctorphilt, 2024-06-09:
Mine must be a VL406U then as it has two buttons for temperature. The instruction manual said it was one or other (attached). I would be very happy to purchase one off you – not sure how we would do this on here, not sure how we are supposed to swap contact details but happy to publish my email address if no other way…

[image]

doctorphilt, 2024-06-30:
My contact is pjturner AT doctorsDOTorgDOTuk in case this helps:)

StarkJohan, 2024-07-12:
I've sent you an email with some info on current state of the PCB.

colsonpete, 2024-07-27:
I am UK based and I’d be very interested too, and happy to pay for it.  I have a VL406T panel (so only 4 buttons: Jets 1, Jets2/Blower, Lights and Temp) connected to a GS515Z.  

I bought MOC of 5 of the MagnusPer/Balboa-GS510SZ board last year (so have 4 spares) but couldn’t get it working as the pumps cycled on and off a couple of times a second.  

I have since realised that I had connected the PCB to the second RJ45 socket on the controller PCB e.g. in parallel to the panel rather than in series with the panel. I have tried it again today (now wired correctly, I think) and it is no longer causing the pumps to cycle, so progress of sorts.

However, I am not getting anything out of MQTT so suspect you are right about the pinouts being different on the VL406 range. 

Please let me know when you have anything ready for testing.

StarkJohan, 2024-07-28:
I have two extra PCBs finalized but the github repo is not ready for publishing yet. I'm caught up with a very big hardware project currently so everything else has been put on hold.

However. While the pinout is the same, the software has to be altered as the actual button functions are not the same. The basic function of reading the data, sending button presses and reading button presses from the panel should work as is.
I might just get a very basic POC project up on github with a general function test as I've been getting some interest in PCBs during the last months.
I'll charge €39 + shipping for a finished PCB at this point. Please send an email to johanATpunktvvvDOTse and we can get the details sorted.

badgeraudio, 2024-08-01:
Another interested party to the new project. 
I have a GS500Z with I think a VL401. I have tracked it down to this so far. 
Pin 1 +5v
Pin 2 lights
Pin 3 Jets
Pin 4 0V
Pin 5 Display
Pin 6 Display
Pin 7 Blower
Pin 8 Temp

Momentary pulling Pin 2/3/7/8 simulates a button press. 
1 button press for temp makes it the display flash, a 2nd press reduces the set temp by one deg. 
Once its stopped flashing (1 or 2 seconds), press it again to make it flash again
Then press temp again and it starts increasing the temp with the button presses. 

I did try powering a wemos mini from the 5v line, and it didnt like it! In that the controller appeared to reboot and the voltage drops when the wemos is connected. 
so we might need to have a separate psu within the spa.

s1dave-web, 2024-08-13:
Another UK citizen here hoping to automate my tub!

I have a 13amp StrongSpa G2 equipped with a GS501SZ and 5 Button VL701S.

@StarkJohan I will contact you via email for more details

adamwylde, 2024-08-28:
> I've ordered a bare PCB that should be compatible with the following controllers and their equivalents: "Mini": VL200, VL240, VL260, VL400, VL401, VL402, VL403, VL404, VL406T, VL406U "Deluxe": VL600S, VL601S, VL700S, VL701S, VL702S
> 
> The pinout, the protocol and the exact functions differ a bit between the two groups. The pinout is easily adjusted with a couple of jumpers in my design. The software needs a slight adjustment for the "deluxe" panels but rather large changes to be compatible with the "mini" models. I've got a POC running nicely on my Mini controller since a few weeks and I will publish a repo as soon as I've had the time to tidy up everything a bit.
> 
> I will be populating a few of the PCBs in the coming month so if anyone is keen on testing/developing on any other panel than VL406U or VL700S, please let me know below. If everything checks out I will be open sourcing the hardware design as well as the altered code I'm using.

I’m very interested in this too. I have a GS500Z unit with VL401 (2 speed pump/lights/temp up/temp down). Are you close to publishing please @StarkJohan? Many thanks

StarkJohan, 2024-08-28:
Outside of work I've been very busy with another hardware project the last month or so but the PCB has been running nicely installed in my own tub for quite some time now. I really intend to get back to this in the next few weeks, I promise! I need the code to be more dynamic for my own install as well so it's just a matter of time. Please be patient and I'll keep you posted.
As it seems, even if the PCB is compatible with a range of different displays, the software will need adjustments to match the different ways the displays work. My plan is to publish a very basic POC and the example I've adapted to my specific panel. Then we'll have to adapt the code to as many displays as possible. Expect some coding and testing!

adamwylde, 2024-08-28:
Great. My son is great with electronics and I’m a software developer (although not in C++ normally!) so very happy to, and capable of getting a PCB ordered from Gerber files, and play with some code. We’ve already done the same for this GS510 project before realising our display is too different to make use of the board.

StarkJohan, 2024-08-28:
[image]
This is the preliminary design, it needs a few small adjustments but it plugs straight into the main pcb.

adamwylde, 2024-09-22:
Hey @StarkJohan got any further this this? 

Still keen to help/beta/get made up, especially as the cold season approaches!

I’d like to be able to manipulate when the hot tub heats via home asssistant to make use of our cheap overnight electricity. 

Many thanks

kgstorm, 2026-01-18:
@StarkJohan @badgeraudio @adamwylde 
Looking through this post, I think you guys have similar/same topside controllers as mine.
I just finished my project.
It's here if you're interested:

https://github.com/kgstorm/Balboa-GS100-with-VL260-topside