MagnusPer/Balboa-GS510SZ #21: CalSpas 407T and CS6215R1A

smitterer, 2024-06-15:
Hi!

I have a Cal Spa CS6215R1A which seems to be some special version of GS500Z with a Cal Spa 407T Display.

Here are some images:
https://photos.app.goo.gl/NoeShParJBgJhQFB9
https://photos.app.goo.gl/DKkjtgdfRPYoWXpB6

Do you have an idea if your project would work with our controller?

Thanks!

mperss54, 2024-06-17:
Hi,
Sorry don't have the knowledge to answer your question about compatibility between CAL SPA control system and the GS510SZ I used as reference during the development.

MagnusPer, 2024-06-17:
Hi,
Sorry but I don't have the knowledge to answer your question about compatibility between CAL SPA control system and the GS510SZ I used as reference during the development.

smitterer, 2024-06-17:
Ok thx.

But ordering the PCB from JLCPCB and connecting it shouldn't be an issue/dangerous due to the same base of the Spa-PCB I guess?

I'd then "just" need to change the code to address the different buttons.

Thanks

MagnusPer, 2024-06-17:
That I can't confirm since I don't have the knowledge of your system