MagnusPer/Balboa-GS510SZ #24: GS501SZ + 5 Button VL701S - No Display Data

s1dave-web, 2024-09-07:
Hi Guys,

I have a StrongSpa G2 with Balboa GS501SZ + 5 Button VL701S top panel.

I have the PCB setup by JLCPCB and connected to the Spa as follows:
DISPLAY-------->(RJ1) PCB (RJ2)--------->(J1)Control system

I am using a standard RJ45 cable that came spare with my router between PCB and Control System

My initial issue is I don't seem to get accurate states or display data:

[image]

Buttons seem to function OK for the most part with Balboa_GS_Read over serial:
1 - OK - Temp goes up
2 - OK - Temp comes down
3 - OK - Mode functions as expected when used with 1/2
4 - OK - Light cycles on/off when sending 4
5 - OK - Pump activates low setting, sending again activates high setting, sending again sets pump to off
6 - NA - No 2nd Pump
7 - NA - No 3rd Pump
8 - Causes a loop to lowest temperature and display will not accept any input
9 - Causes a loop to lowest temperature and display will not accept any input

So it would appear most buttons function as expected other than 8/9 which should be fine for my needs - any help on the display data?

Also when using Balboa_GS_HomeAssistant the light button doesn't seem to work but is perfect over serial with Balboa_GS_Read

netmindz, 2024-09-08:
Looks like I made an error on this line. It should be lights plural 

https://github.com/MagnusPer/Balboa-GS510SZ/blob/46f41fbbb927061c465f90602a6bfaaeec2079b3/examples/Balboa_GS_HomeAssistant/Balboa_GS_HomeAssistant.ino#L260

s1dave-web, 2024-09-08:
Buttons function as expected now thank you @netmindz 

Balboa_GS_HomeAssistant had the following adjustments:
Line 252 change if (s_payload == "TempUp") { to if (s_payload == "Temp Up") {
Line 256 change else if (s_payload == "TempDown") { to else if (s_payload == "Temp Down") {
Line 260 change else if (s_payload == "Light") {  to else if (s_payload == "Lights") {

Now just need to work out how to get display, heater, light, pump and water temp data

netmindz, 2024-09-08:
Glad you sorted it, do you know how to create a PR to submit your fix? 

As for displaying the state, that in theory should already be sent to HA

Does HA not show the current state of the lights if you toggle on the tub for example?

s1dave-web, 2024-09-08:
I think I have submitted a PR now after watching a YT tutorial...

HA is receiving the state data, its just incorrect (everything shows in ON state and does not change, display shows 88.88 and Water temp 0c)

I can only think that maybe the GS501SZ differs to the GS510SZ when sending state data? Or possibly my read & clock pins are not getting a proper connection... I will inspect the board for continuity tomorrow but suspect it is OK as panel on tub is 100% functional.

netmindz, 2024-09-08:
Yeah if the values are wrong when looking at the serial output, then not an issue with the home assistant part so I'll stand down

s1dave-web, 2024-09-09:
Great progress today - A new Mini D1 fitted and I now have display data. I will close this issue and open a new one for the last remaining issue - HVAC controls.

closed 2024-09-09 by s1dave-web