MagnusPer/Balboa-GS510SZ #26: HA HVAC control - unable to change Target Temperature

s1dave-web, 2024-09-09:
With a new D1 mini I now have a fully working setup with sensor and display data:
[image]

The only part not functioning is HVAC - so far I have managed to get it to have the correct current and target temperature by changing line 222:
deleted:
          hvac.setCurrentTargetTemperature(Balboa.waterTemperature);

added:
          hvac.setCurrentTemperature(Balboa.waterTemperature);
          hvac.setTargetTemperature(Balboa.setTemperature);

Ultimate goal would be to get temperature changes on the HVAC card to increase the setTemperature value.
[image]

I have a feeling something with onTargetTemperatureCommand should be changing temps:
void onTargetTemperatureCommand(HANumeric temperature, HAHVAC* sender) {
    float temperatureFloat = temperature.toFloat();

    Serial.print("Target Temperature: ");
    Serial.println(temperatureFloat);

    Balboa.updateTemperature(temperatureFloat);

    sender->setTargetTemperature(temperature); // report target temperature back to the HA panel
}

But I am a bit stuck... hoping someone with ArduinoHA knowledge could help... @netmindz possibly?

netmindz, 2024-09-09:
The changes look correct, if you logger is showing the right target value when you change, then it's an issue with the set temp call in the library

s1dave-web, 2024-09-10:
Serial monitor shows nothing when changing the temp on the HVAC card

Using Balboa_GS_Read.ino sending case 8 and 9 work fine... so it would appear HA isn't sending command

s1dave-web, 2024-09-10:
Is there a simpler way without using HVAC card that I can set the temperature to a specific value within HA? At the moment the only way would be to send Temp Up a certain amount of times.

netmindz, 2024-09-10:
It's missing

hvac.onTargetTemperatureCommand(onTargetTemperatureCommand);

s1dave-web, 2024-09-10:
That was it... so silly of me!

I am going to make a PR with a new file for 1 pump tubs because the new version of the current file has changed to switches which does not work for me as I need to be able to press pump1 twice to alternate speeds so button works best.

closed 2024-09-10 by s1dave-web