MagnusPer/Balboa-GS510SZ #30: ESP yaml help

Doser86, 2024-10-03:
I am unable to install the Balboa_GS_HomeAssistant.ino
Getting a lot of errors in the yaml.

netmindz, 2024-10-03:
The file is a sketch you build with the ArduinoIDE and populate with the connection details to the MQTT server that is attached to your Home Assistant setup

You make no modification to any yaml to use this code

Doser86, 2024-10-03:
Aaaahhh. Sorry. I am only used to flashing with ESPhome. 
So only flashing this file with Arduino IDE and nothing more? Mqtt server i allready have.

netmindz, 2024-10-03:
No worries. Yeah just need to set the right connection details before you flash

Doser86, 2024-10-03:
Thankyou :)

I messure different in my RJ45. pin 1 is about 5v. Pin 4 i ground. How do I messure wich is data kables?

Doser86, 2024-10-04:
I get no readings. Tried every combination with wires. Do I have to have the board you are using? I installed the code on a Atom M5-stack.

netmindz, 2024-10-04:
Are you sure your controller is the GS510SZ ?

Doser86, 2024-10-04:
Is this on the cover of the pump? I have a GS, that I know. Inside the cover is saying MS500ZE - PN 56861

netmindz, 2024-10-04:
So that is a different model, you would need to research if the topside controls for the GS500Z are also compatible with the GS510SZ

netmindz, 2024-10-04:
The GS510 says "Topside controls from the serial standard VL range such as the VL700S and the VL701S"

The GS500 says "It pairs excellently with topside controls from the duplex VL range, such as the VL404 and MVP260. However, it’s most commonly used with the Mini Panel VL200."

So different controls so this library won't be compatible

Doser86, 2024-10-05:
But this should just be finding the data from VL200 and change the include files right?

netmindz, 2024-10-05:
My point is that this code won't work with your tub.
Your task is to understand what the differences are between the two systems and then create new code, probably using this as your starting point, but having it's own feature set.
As this is the same manufacturer, it's most likely their own code is a respin of what they have done before, you just need to find out what extra capabilities and changes they made with your board. 
At the end of the day we are emulating the topside controls, not the controller

Doser86, 2024-10-05:
This is exactly what I was thinking. How did you find out yours is working?

netmindz, 2024-10-05:
My own tub uses a totally different communication method, so I had to reverse engineer their serial protocol

Doser86, 2024-10-06:
This is a little bit above my knowledge I think

Doser86, 2024-10-08:
My board is saying: When J12 on the Main Board is jumpered on pins 2 & 3, the system is in GS50xZ compatibly mode.

If J12 on the Main Board is jumpered on pins 1& 2, the system is in GS51xZ/GS5xxSZ/GS5xxDZ compatibly mode. 

Maybe solution is here?

netmindz, 2024-10-08:
Worth a try, however you may well then use some/all use of your current topside controls, which may or may not be an issue. In the worse case you might apply the wrong power to the wrong pins and cause damage to your existing topside control, so best to check exactly what that change of setting does without the topside panel attached

Doser86, 2024-10-08:
I try tonight without the top panel connected. If this works then I could just buy a new panel than works or make a switch to switch between HA and top panel

Doser86, 2024-10-09:
Should I have both Balboa_GS_Interface.cpp and Balboa_GS:Interface.h in the sketch map? I am not able to flash when I have them in the same folder.

netmindz, 2024-10-09:
This is a library. You need to add  the entire project into the library folder for your identity, then open one of the examples from the examples menu in the ide

Doser86, 2024-10-09:
Sorry for stupid questions. Is there like one zip folder that has ewerything in it?

netmindz, 2024-10-09:
https://github.com/MagnusPer/Balboa-GS510SZ/archive/refs/heads/main.zip

Doser86, 2024-10-09:
So I "Include library" - "Add zip.library" - then go to exampels and open lets say Balboa_GS_HomeAssistant. Then just change my wifi, server and mqtt?

First time using Arduino IDE

Doser86, 2024-10-09:
Nothing more? Then just flash?

Doser86, 2024-10-16:
Your wiring thru EL816... Is this necasarry? Or kan I just go straight from Balboa read and write wires an in to D2 and D8?
How did you messure wich wires this was on your tub?

s1dave-web, 2026-11-09:
Just to let you know I made a ESPHome component for this excellent tool - it is coded for my specific tub (5 button panel - 1 pump) so is classed as experimental for any 2/3 pump tubs but I added some smarts for a mode select, pump speed detection, filter duration detection, filter cycle prediction (for Eco and Sleep modes), time to heat calculations:

The pump speed is guessed from button press history + an optional energy sensor to detect if someone presses on the panel and the pump goes from LOW/HIGH - I havent extensively tested any of this (cold months now) but initial functionality seems fine:

https://github.com/s1dave-web/Balboa-GS510SZ-ESPHome