MagnusPer/Balboa-GS510SZ #33: GS500Z with Balboa-VL260-PN-55081 control panel

surfmk, 2025-06-06:
Hi all,

This is a great project! I have a Cove Spa Delux that came with my house and I'm getting into home automation.

Before I jump into this and get my soldering iron out, could anyone comment on:

1. Code compatibility with GS500Z and my 4 button VL260?
    I'm happy to do some code editing and have a hobbie background in C++

2. I have a LAN connection available I would like to use in preference to Wemos D1 mini
    (powerline LAN allows extra distance into garden - out of wireless range)

I attach some photos of my actual hardware...

Many thanks in advance,

Paul

[image]

[image]

[image]

[image]

MagnusPer, 2025-06-07:
Hi @surfmk 
Nice you visit this GitHub project! I can not answer the question on compatibility for your GS500Z control system since I reversed engineered GS510SZ. Either you need to do the same approach I did to measure the link between control system and display using oscilloscope or just give it a try and hope for the best!

Jacktheripper125, 2025-06-26:
Hi, I can confirm it doesnt work sadly. It stops the pump button from working and no useful data is gained.

surfmk, 2025-06-26:
Hi Jack, That's interesting you are following this up... I haven't started yet but am happy to see what protocols are being used when I do. I'd quite to find a remote panel (chances are remote!) and see if we can replicate the signals that this sends...
All good fun.
Paul

netmindz, 2025-07-03:
You might find a logic analyser is more helpful than a scope

Jacktheripper125, 2025-07-03:
Thank you netminds.

One thing i would say is dont try the software and card on a gs500. After usung it my hot tub went to 40c and stayed there even though i selected 34c it just kept heating to 40c after haveing the card plugged in. I tried re powering and re setting. Even tried a new temp sensor. I have now had the tub un plugged a few days and the memory wipe jumper on and touch wood today it is back to normal. 
I will maybe wait until i get brave again. I will get a logic board as it looks interesting. I will find something cheaper to break first

surfmk, 2025-07-21:
Hi Jack, just wondering if any more progress? A logic analyser seems to be the way forward as netmindz suggested. I'll try to learn a bit more about how these work as I've only ever used an oscilloscope. If we can work out the protocol, we should be able to see where it diverges from the GS510*Z.

kgstorm, 2026-01-18:
@surfmk Were using the same topside controller, so my project should work for you to.

Have a look if interested:

https://github.com/kgstorm/Balboa-GS100-with-VL260-topside

ETE-Design, 2026-04-25:
@kgstorm This is mine board - Looks like it match really much as the thread starters pictures.

[image]
[image]

kgstorm, 2026-04-25:
@ETE-Design  The ESP32 C6 is single core. I initially started my project with an ESP8266 and I could not get it to work and I think single core was likely the problem. Since my project uses Wi-Fi, but also very reliant on a lot of CPU for all of that timing to capture the data, I think you will struggle with an ESP32-C6

GevaudanBeast, 2026-04-25:
## ✅ Full working integration - GS501Z R2B + VL403 + ESPHome + Home Assistant

Hi everyone,
I posted earlier in this thread about my findings on the GS501Z R2B. I can now report a **fully working integration** with complete read AND write (button simulation) via Home Assistant. Here's everything I learned.

---

### Hardware
- **Mainboard**: Balboa GS501Z R2B (firmware v43)
- **Topside panel**: VL403 (4 piezo buttons: Blower · Jets · Temp · Light)
- **MCU**: ESP8266 NodeMCU v2 (large format)
- **RS485 reader**: TTL485 module on J18 (TX only, read-only - confirmed)
- **Button simulator**: HY-M154 PC817 4-channel optocoupler module on J1 or J2 as you wich !
- **Step-down**: adjustable module 14V → 5V on VIN pin of NodeMCU

---

### RS485 reading (J18) - passive telemetry

As confirmed in my earlier post, J18 emits clean **27-byte frames** at ~5 Hz:
```
64 3F 2B [data...]
```

Key bytes decoded after extensive capture sessions:

| Byte | Content |
|------|---------|
| b3 | Water temperature (× 0.5°C, rounded) |
| b4 | Heater outlet temperature (× 0.5°C) - sensor B |
| b5 | Set temperature (× 0.5°C, rounded) |
| b9 | Filter cycle duration: 0x02=F2h, 0x04=F4h, 0x06=F6h, 0x08=F8h, 0x0E=FC |
| b17 | Pump state (0x00=OFF, 0x01/0x08=LOW, 0x02/0x18=HIGH) + bit7=blower |
| b18 | Mode SL flag (0x08) + light bits (bits 1-2) |
| b19 | Heater ON (bit0) |
| b23 | Mode: 0x20=ST, 0x40=SL, 0x00=ECO |

---

### Button simulation (J1) - write control

The VL403 uses **piezo buttons** connected to J1 via RJ45 (T568B). The bus carries:
- **pin1** (White/Orange): Common +5V 
- **pin2** (Orange): Light button
- **pin3** (White/Green): Jets/Pump button  
- **pin7** (White/Brown): Blower button
- **pin8** (Brown): Temp button
- **pin4** (Green): GND

Simulating a button press = shorting the button pin to pin1 (+5V common) for ~100ms.

**Optocoupler wiring (HY-M154 PC817 4-channel):**
- Remove all 4 red jumpers
- Power the module: 5V from NodeMCU VIN pin → all 4 IN terminals tied together
- GPIO ESP → IN per channel (D5=Blower, D6=Pump, D7=Temp, D0=Light)
- GND ESP → G (input ground)
- V1 × 4 → pin1 J1 (+5V common)
- G output × 4 → each button pin (pin7/pin3/pin8/pin2)

⚠️ **Critical**: The ESP8266 GPIO outputs ~3.1V which is insufficient for the PC817 LED. You must power the module's IN side at **5V** (NodeMCU VIN when powered via step-down set to 5V). Feeding 5V to VIN, the NodeMCU regulates internally to 3.3V for the MCU while exposing 5V on VIN for the optocoupler module.

---

### VL403 button combinations

The VL403 uses **sequential** (not simultaneous) button presses for special functions:
- **Mode change** (ST→ECO→SL→ST): press Temp, then 400ms later press Light
- **Filter cycle programming**: press Temp, then press Jets to enter programming; press Temp to cycle between F2/F4/F6/F8/FC; press Jets to exit

---

### ESPHome firmware (ESP8266)

Full YAML firmware available - key points:
- UART on GPIO4 (RX from TTL485 TXD pin), 9600 baud
- Parsing ASCII HEX frames between `[` and `]` brackets (the TTL485 module outputs ASCII, not binary)
- 500ms interval (critical - 200ms causes Hardware WDT crash on ESP8266)
- Timings: 100ms button press, 400ms gap for combos
- Native ESPHome API (no MQTT needed)

---

### Home Assistant integration

- Full `climate`-like control via package YAML (input_number + input_select + scripts)
- Scripts handle mode cycling (always via ST as safe position) and temperature changes (circular 26→40→26)
- Debounce 3s on temperature slider
- Notifications for disconnection and diagnostic errors
- Filter cycle displayed as F2/F4/F6/F8/FC
- Delta temperature sensor (heater outlet - basin) for circulation monitoring

---

### Key lessons learned

1. **J18 is TX-only** - confirmed, no injection possible
2. **J1 is NOT RS-485** - it's a proprietary 5V synchronous bus (SPI-like) for the display; the button lines are separate passive lines
3. **Piezo buttons work with contact closure** - any optocoupler or relay works
4. **3.3V GPIO is insufficient for PC817** - need 5V on the LED anode side
5. **Filter duration is in b9**, not b23 (b23 encodes mode + oscillating flags)

Happy to share the complete ESPHome YAML and HA package if anyone is interested. This approach should work on any GS500Z/GS501Z with VL400-series panels.

ETE-Design, 2026-04-26:
@GevaudanBeast If I read your post correct we need both TTL485 module and HY-M154 PC817 4-channel optocoupler module?
And we can use J2 to connect optocoupler?

While I see you have use an ESP8266 NodeMCU v2 I have some SeedStudio ESP32-C6 lading around so would like to use - You Think it is possible?

I really like how @kgstorm Custom Card look - Would it be possible to fetch it to work with your integration?

Any chance you can make a connection diagram?

Pls. Share YAML and HA Package

ETE-Design, 2026-05-17:
@kgstorm 

The description from the manual:

**TEMPERATURE DISPLAY AND ADJUSTMENT**
Factory Default Settings:
• Temperature: 37°C
The LCD window displays the actual water temperature. Pressing the TEMP+-
pad makes the numbers flash, then pressing the pad again (repeatedly)
increases the set temperature one degree at a time, up to 40°C. After a short
pause, the actual temperature displays. Press TEMP+- pad repeatedly to
decrease the set temperature to 27°C.